Shaztech/Pilldispenser #5: Green LED stays one after dispensing

jsondag, 2024-04-20:
When dispensing a tray.
If you dismiss the "Take your Pills" screen, 

at 30s, the LED comes back on green, and stays green till four minutes after dismiss. This appears to be two times the 120s screen timeout.
00:00 Dispense then dismiss.
00:30 Green LED comes on
02:00 LCD turns off.
04:00 Green LED turns off

If you don't dismiss it and let it time out.
00:00 Dispense
00:30 Take you pills goes away
02:00 Green LED comes on
04:00 Green LED turns off.

Shaztech, 2024-04-20:
Does your tray is configured for green ?

jsondag, 2024-04-21:
It’s configured for red. I’ll check the voltage when I get a moment. Seems an odd one that it happens in line with the screen timeout of 120S. It would also be nice to have a configurable timeout and a button (perhaps the image of the display on the base) that would immediately shut off the screen and LED. I was studying your code to see if I could add those things, but I’m not the best coder, so I don’t yet quite understand it.

Shaztech, 2024-04-21:
I have timers that do things in those times. And sends a command to put the LED "off" but to put an LED "off" is to send the color "0,0,0" in HEX therefore black. If there is not enough voltage the chip in the LED reacts strangely and displays green.

You want a configurable timeout/button to do what ? The LED is supposed to always be off except when there's an alert and if the screen is in "dispense". (Also a small bluish tint when it's booting)

Mine with the exact same code posted here, is working fine for over a month (multiple users and alerts per day).

Does your LED flash red (or the color of your tray) during an alert/dispense ?

jsondag, 2024-04-21:
[image]

jsondag, 2024-04-21:
A configurable timeout for the screen, and/or button to blank the screen. I tend to take my meds and then lay right down to sleep while my wife does the same. Two minutes with the screen on may annoy her.

jsondag, 2024-04-21:
https://github.com/Shaztech/Pilldispenser/assets/13071956/d11f448d-127b-478c-bead-82c6e69a5b71

jsondag, 2024-04-21:
It’s just powered from a PC at the moment. So that could be the problem.

Shaztech, 2024-04-21:
If you get a good 5.0V at the LED module, maybe your WS2812B is faulty, they are really sensitive (or cheaply made), so the heat when you soldered it can have killed it.

I'm thinking of where to put the timeout choice (limited space in the setting menu), or if I simply change it to 15 sec when the mute button is active. Since the mute is usally used when you are in an environnement where you don't want to disturb.

Shaztech, 2024-04-21:
Also from your video I can see that when it's flashing it's amber (that's green + red) so the green is stuck.

jsondag, 2024-04-21:
Multiple setting pages maybe?  There are likely other things that could be added down the road. Also, a page to show IP would be good. You currently only see it when you connect. I had to check DHCP leases on my router to find it again.

Shaztech, 2024-04-21:
When you start it, it's showing on the loading screen ;)

Shaztech, 2024-04-22:
I will add a button to turn off the screen manually
[image]

Shaztech, 2024-04-23:
Try this
[PillDispenser.esp32.v1_4.zip](https://github.com/Shaztech/Pilldispenser/files/15070019/PillDispenser.esp32.v1_4.zip)

mrperson2015, 2025-01-12:
I'm experiencing the same issue. The LED remains green constantly, only turning off a few seconds after the screen turns off.

This is happening with version 1.8, even after performing a factory reset. I've also tried a fresh installation using esptool. To rule out hardware issues, I tested another board I had set aside for a new dispenser. After loading it with the same software, the LED on the back behaves the same way.

The voltage to the LED board (from jsondag) measures 4.71V.

Shaztech, 2025-01-13:
I get the same problem, sometimes the LED turns green for no reason. I've tried to change some things in the code (before version 1.8) and still the problem persist.
I don't know how to fix this. 
If someone want to help, that would be great.

mrperson2015, 2025-01-13:
Perhaps @Shaztech can provide some guidance on remotely debugging the code. I'm new to microcontrollers but have experience with general coding and remote debugging. I’ve already set up the project and have it running in Arduino IDE 1. I can write and upload code successfully, but I need assistance with the debugging process.

Shaztech, 2025-01-13:
Check everywhere "pixels" is called. 
I'm using the library Adafruit_NeoPixel.

It's also normal that every change of the LED is made on 0 and 1
Ex:   
pixels.setPixelColor(0, pixels.Color(213, 230, 255));
pixels.setPixelColor(1, pixels.Color(213, 230, 255));

Because if you don't use the small driver board, you can use 2 LED Chip, the first one as a sacrificial LED.

mrperson2015, 2025-01-13:
@Shaztech, is there any way of enabling logging?

mrperson2015, 2025-01-13:
I have the serial debugging in place now. I can trace things now. Will take some time complete. I will report back w/ my findings. Thanks again for the project and the help.

Shaztech, 2025-01-14:
I think it's an hardware issue or a bug in the library version

jsondag, 2025-01-14:
I had this issue initially, but after doing an ota update it stopped happening. Though that was several releases ago.

mrperson2015, 2025-01-14:
It is happening even after I erase the flash and load from cmd using the release files. I suspect it is hardware related. I dont know enough about electronics to test the led board for errors. I cant imagine it is happening on 2 different ESP32 boards and causing the issue.

Shaztech, 2025-01-15:
Can you try to power the LED board with an external 5.0 to 5.1v
Too see if the voltage is too low.

jsondag, 2025-01-15:
If you power the board from an external supply, make sure the ground of your esp AND the ground of the supply are tied together. Then make sure only your supply positive is tied to the positive of the led board.

I would first measure the voltage at the ground and positive pins directly on the led board, if it's 5V then testing with an external supply would be unnecessary.

Technically minimum supply voltage for the chip on that board is 4.5V, but it's possible you have a bad chip. If you got it from me, I'll replace it, if it's bad.

When I had this behavior, my voltages were fine. I haven't watched closely to see if this happened again, as I usually walk away after dispensing, but I'll watch for it over the next few days.

mrperson2015, 2025-01-15:
I get anywhere between 4.6-4.71v measured from the pins on led board as powered from the esp32. I removed the positive pin from the board and added a positive and negative to the other side from an external 5.1v power supply. It acts the same. Intermittent behavior on the greenish light. darn it. @jsondag, I will be ordering a new board. I will send you an email. 

Thanks for all your help.

mrperson2015, 2025-01-18:
We should close this issue unless you want to wait to verify against a new LED board.

closed 2025-02-03 by Shaztech