Shaztech/Pilldispenser #6: Servos

bnred777, 2024-04-20:
Hello, I purchased some servos from aliexpress that were supposed to be towerpro sg90 servos. Seems they might not be exactly the same as to what you have used. I can't seem to install the server because it is a bit too tall. I measure about 1.5 to 2 mm of gap between the servo and tray mount face where the screws go. 

Could you please provide the measurement of the servos that you actually used as described here in measurement "F" in the following link? https://www.towerpro.com.tw/product/sg90-analog/

Also, would it be possible to get a link to where you purchased for servos the project?

Thank you in advance, I'm really looking forward to completing the project. Excellent work!!!

Shaztech, 2024-04-20:
I've had this issue on some servos I got from Aliexpress, you can use them "as is" without screwing too much. The new arm (2 weeks ago) accomodate for the shorter/taller servos.

closed 2024-04-20 by Shaztech

bnred777, 2024-04-20:
Thank you! After looking at your images I believe you are using analog servos correct?

Shaztech, 2024-04-20:
I don't understand you question.

I based my design on SG90 servo model I found online (grabcad) and it was the same with some I had here (old stock). So I don't know if there was a revision or it's just different from any manufacturers. I've seen some picture of another user tray and it was fitting perfectly.

bnred777, 2024-04-20:
Oh I see, tower pro supplies a digital version of the SG90 as can be see on their website here: 

https://www.towerpro.com.tw/product-category/servos-parts/micro-servo/

I too am not 100% of he difference but I would assume one is designed for serial communication vs a standard analog version that uses voltage to determine where it need to position. I believe you are using PWM in your code so that would mean the analog version is the correct one (I think at least)

Shaztech, 2024-04-20:
Yes I use the "standard" PWM (Analog) version. The one you can buy for 2$ everywhere XD

bnred777, 2024-04-20:
Perfect, thank you again!