Shaztech/Pilldispenser #7: How many Telgram users ...

RoboCop0815, 2024-04-26:
...
can I set up and how?
is only one possible or can I save several tokens one after the other?
Greetings Robbie

Shaztech, 2024-04-26:
Only one token and one chat ID.

Here's some snippet from https://core.telegram.org/bots/tutorial

**Obtain Your Bot Token**
Obtaining a token is as simple as contacting [@BotFather](https://t.me/botfather), issuing the /newbot command and following the steps until you're given a new token. You can find a step-by-step guide [here](https://core.telegram.org/bots/features#creating-a-new-bot).

Your token will look something like this:

4839574812:AAFD39kkdpWt3ywyRZergyOLMaJhac60qc
Make sure to save your token in a secure place, treat it like a password and don't share it with anyone.

**Get your chat ID**
Talk to your new bot inside telegram and 
go to https://api.telegram.org/bot<YourBOTToken>/getUpdates
You will see the last message and search for "chat":{"id":
the digits will be your chat ID (can include a minus in front of the digits).

RoboCop0815, 2024-04-26:
It wouldn't be bad if several users (several mobile phones) could also receive their own Telegram on these... that would be great...

Shaztech, 2024-04-26:
It's a limitation of Arduino with telegram. But the chat id can be a group chat with multiple users. I already tell which tray with the description in the alert message.

closed 2024-04-28 by Shaztech