Shaztech/Pilldispenser #8: No LED display

RoboCop0815, 2024-04-29:
In my case, the LED shows nothing, neither RED nor green or anything else.
But the 74AHCT125N gets very warm (about 70°C/150°F), is that normal?

Where is my error?

Shaztech, 2024-04-29:
The IC is not a 74AHCT125N but a ATTiny414.
The orientation of the IC seems ok, try a new board with new components.
You can also try with a strip of WS2812B of 2 leds (without the board and the 75A...) and cover the first led with black tape.
This board is only to raise the signal level (data in) from 3.3v to 5v

RoboCop0815, 2024-04-29:
thanks, I have tried it with the 2 x WS and the LEDs do their colors as it should, only now I have no more alert... Is that normal?
or have I torn off a cable, lol

Shaztech, 2024-04-29:
The first LED can sometimes act weird (lower voltage on the data pin) that's why I said to cover it with a black tape.

RoboCop0815, 2024-04-29:
I just put them in a heat shrink tube..lol

Shaztech, 2024-04-29:
That will do better than the tape !

closed 2024-04-29 by Shaztech