Shaztech/Pilldispenser #12: Detailed Compile instructions with Arduino IDE

ibgregr, 2024-05-24:
So I set up the Arduino IDE to use the "ESP Dev Board" as everywhere I read it said to use that (or any other ESP variant).  But when I try to compile the Sketch I get the following:

Sketch uses 1691373 bytes (129%) of program storage space. Maximum is 1310720 bytes.
Global variables use 116384 bytes (35%) of dynamic memory, leaving 211296 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Compilation error: text section exceeds available space in board

I'm not sure if I have the incorrect board settings or what.  I tinkered with the "Partition settings" and got it to compile but I'm not sure I even chose the right entry there either.  I would like to have some more detailed steps on how to compile and directly upload to the board from the IDE (if it's even possible).  If not, the steps to generate the bin files like you have to upload with esptool.  Nothing I have tried has worked and I end up having to just reflash your bin files.  It's been YEARS since I have messed with any Arduino type stuff and my old brain has purged all of that knowledge.

Also, the readme section about loading the bin files references a file called boot_app0.bin which is not one of the bin files you have.  And the esptool command does not reference all 5 bin files that were downloaded.  Can you clarify exactly what should be used to reflash the board?

So..to sum it up.  I would like to know if it's possible to compile and upload the app from the Arduino IDE...and if so...what are the board settings, etc I need to use.  If that's not possible, can you give detailed steps on how to produce the bin files and the detailed steps to upload it.

Thanks for such an awesome project!!

GregR

Shaztech, 2024-05-24:
[image]

Use the minimal spiff setting under partition scheme !

[image]

Also if you want to create a bin file use the "Export compiled binary"

Shaztech, 2024-05-24:
for the spiff data file you need to click "ESP32 Sketch data upload", the file will be in the temp directory.
[image]

If the light turn on a few seconds it means the code is running, only the LCD isn't showing.
Did you copy all the library files into your documents ? 
Because there's a config file in the TFT_eSPI that is special for this LCD module.

ibgregr, 2024-05-24:
Yes..I copied the libraries and PillDispenser folders from your Pilldispenser-1.5.zip file to my Documents\Arduino directory.  The libraries folder has the lv_conf.h file which is what I think you are referencing above.  So I should have exactly what you have in the zip file.  This compiles just fine and uploads but again the LCD never comes on.  Just the LED for a bit and then it goes off as well.  Are you sure the zip file is as expected?  Also, I do not have the "ESP32 Sketch Data Upload" menu item.

Something is not quite right here...

ibgregr, 2024-05-24:
Ok...I'm using the v2.x of the Arduino IDE and have found the Upload ESP32 Sketch Data may not work...but...I have found the tool to recreate the SPIFF files so all is good there.

As for the other files.  I have EXACTLY what came from your PillDispenser-1.5.zip file.  You state you included all of the libraries you are using but I have found that the UniversalTelegram library is not in the libraries directory and had to be installed so I'm wondering if the zip file is complete?  I'm assuming I should be able to copy the PillDispenser and libraries folders to my Documents\Arduino folder and compile/upload this to the board and it should work.  But that is not the case.  The code compiles and uploads but the LCD display does not come on.  Any other ideas?

ibgregr, 2024-05-24:
It might help to list the version of each library you are using so that I get the same ones installed.  I say that because it looks like some of this is version dependent.  For example, with the latest version of lvgl installed the code will not compile at all.  And, there's an include in the Adafruit_PWM_Servo_Driver_Library for "#include <Adafruit_I2CDevice.h>" which cannot be found but if I upgrade that library then it is good...but then that requires another library that it installs.  

Anyway...I will wait to hear back on what you find.  I'm thinking it's as simple as a missing file, out of date file, etc that is causing this which means the source zip file is not in sync with your build system.

ibgregr, 2024-05-24:
A bit more troubleshooting I've done.  

1. I copy the PillDispenser and libraries folders to my Documents\Arduino folder and click upload, the compile actually fails due to missing UniversalTelegramBot library and missing files for the BusIO library which is used by the Adafruit_PWD_Servo_Driver Libary.

2. I install the UniversalTelegramBot library (latest version) and the Adafruid_BusIO library (also latest version) then I click upload.  The compile completes successfully, and then the code is uploaded to the board and it reboots.  The LED comes on white and the LCD display flashes when it is updating the strings.  I can see it flash for each of the "Servo lock" messages that would have been displayed but then it ends up off.  So this has to be something that is not configured correctly in the source files provided or it could be a library version issue with the Adafruit_BusIO library that I had to install.

Maybe it would be better to just recreate the source files directly from the system where you built the bin files for v1.5.  Or if you have access to a clean system then you could try copying those folders and see if you can build and upload it to a board and have it work properly.

Thanks again for helping with this.  I just bought you 3 coffees as well!

Shaztech, 2024-05-24:
I use arduino 1.8.19 (I really do not like the version 2.x).
Here's my FULL library (there's stuff you don't need, but it's for testing), I can compile without any problem with this and it work fine.

https://1drv.ms/u/s!AhrI-O28GqX7ia0-orkJlkawpzfHiQ?e=N0E8ff

ibgregr, 2024-05-24:
Ok...I tried that first with the IDE 2.x.  Same results.  I then uninstalled the 2.x and installed the 1.8.  Same results.  Then I totally uninstalled the Arduino "stuff" and removed all directories.  Reinstalled IDE 1.8 and it worked!  Then I tried to install 2.x and it did not work (screen flashes, etc).  Uninstalled 2.x and went back to 1.8 and it is still working.  I removed all of the unneeded libraries for now.  My next test later tonight will be to take your 1.5 zip file and see if that works.  I would really like to figure out why 2.x code does not.

Shaztech, 2024-05-24:
Like I said, I don't like 2.x, the interface is weird for me, simpler with 1.8.x. The constant "updates" it want to do with the libraries (I use older versions sometimes because newer break stuff). So I cannot help with 2.x. But I know that you can use a Portable version of 1.8.x and it doesn't mix libraries (libraries stays in the portable folder).

ibgregr, 2024-05-25:
First of all...I'm not sure what was going on with the libraries.  I still think something is slightly different between what was in the 1.5 zip file and the libraries your local library.  As I stated above, I cleared out everything from that library folder except for the ones required by PillDispenser.  I also have this working fine with IDE 2.x.  I believe some of the issues I was seeing was due to the way things are cached.  Once I totally removed ALL remnants if both Arduino IDE's (and any leftover directories) and reinstalled 2.x it worked...but only when I used the libraries from your local copy.  So that is what I'm going to stick with for now.  I may try to do some kind of folder compare to see if I can find what is different from the 1.5 zip and. your local libraries...but that's a task for another day.  Once again, thanks for sticking with me on this.  I understand you like the 1.8 IDE but it just feels too dated for me after doing most of my coding in Visual Studio (C#, etc).  I agree with you about the annoying popup about library "updates" every time I start the IDE...but that's a small price to pay for progress...I guess. :)

ibgregr, 2024-05-25:
Guess I should have search before I mentioned the "updates" above.  This site tells you how to disable that check on startup so that it does not pop up:

https://forum.arduino.cc/t/i-need-to-prevent-arduino-ide-2-x-from-checking-for-library-updates/1190545

Shaztech, 2024-05-25:
Thanks, will check.
I've updated the libraries folder, added Universaltelegram.
You can check if they are ok

ibgregr, 2024-05-25:
If you want the libraries to be complete you also need to include Adafruit_BusIO and ArduinoJson.  The first one is a dependency for the Adafruit_PWM_Servo_Driver_Library and the second is a dependency for UniversalTelegramBot.  If you add those then you will have everything.

Shaztech, 2024-05-25:
Done

closed 2024-05-25 by Shaztech

ibgregr, 2024-05-25:
Ok..just replaced my libraries folder with a clone from here and it compiles, loads, and runs just fine.  I guess you would need to udpate the tar and zip files that are on the binaries download page so that they have the correct libs as well.  Thanks for getting all of this cleaned up and synced!  I also found a plugin for IDE 2.x that provides the sketch data upload as well.  It just goes out and runs mkspiffs to create the image and then uploads it to the board.  So I have everything working with IDE 2.x now.  Have a great weekend!

ibgregr, 2024-05-25:
One more thing...did you use Squareline Studio to build the UI?  If so can you share the .spj file(s).  I would like to tweak a few things for my own liking without having to figure it all just by looking at the code.

Shaztech, 2024-05-25:
I think you remove your last question, so you probably just found it ?

ibgregr, 2024-05-25:
Yep...as soon as I clicked the button I saw it in the repo :)

jsondag, 2024-11-22:
Hello. Trying to make changes for a version of the software for testing out servo boards. I have my test jig set up, so I can place a board on pogo pins, and use it with modified firmware that actuates all the servos, and then runs the LED through all the colours, when I press the screen off button. 

I was looking into changing the interface a bit to have buttons for each servo on the main screen as well is individual buttons to test all servos, the LED board, and play either file 1 or 2 on the dfplayer.

How do you prepare the asset files for upload? I'm just starting to play around in squareline studio.

Shaztech, 2024-11-22:
[image]

After you modified the Squareline project click on Export UI Files, it will create an arduino folder. Replace the files in the "C:\Users\user\Documents\Arduino\libraries" Probably the folder "ui"

Also, in the Arduino files, you need to add your button event (in the PillDispenser.ino):
ex:  lv_obj_add_event_cb(ui_VolTestBTN, VolTestBTN_event_handler, LV_EVENT_ALL, NULL);
but replace the name with the button name

and the void that do the action (in the LVGL_Buttons.ino):
ex:
static void VolTestBTN_event_handler(lv_event_t * e) { // Test Volume Button
  lv_event_code_t code = lv_event_get_code(e);
  if (code == LV_EVENT_CLICKED) {
    spkvolume = lv_slider_get_value(ui_VolSlider);
    Playsound(2);
  }
}