Shaztech/Pilldispenser #14: Automatic pill throw out by alarm?

panthermatrix, 2024-08-25:
The automatic pill dispenser is good project. With the title, however, I assumed that the pills would or could be ejected automatically, not just an alarm sounded and the throw out had to be confirmed manually on the display.

I would be happy about a function or setting, also optional as far as I am concerned, that gives the alarm, but also throws out the pills of the respective compartment directly. In addition, it should not be necessary to end the alarm by touch with automatic ejection.

Would that be feasible?

burkby, 2024-09-05:
I would also like to have this option.

I have a father-in-law with Parkinson's and Dementia, so I am not sure if he can press the touchscreen to dispense. 

I can see some drawbacks of it. My father-in-law takes meds four times a day. Once with breakfast, once after, in the evening and before bed. I could see a situation where he'd easily combine doses.

Maybe, adding to the idea of auto-dropping, we could do something remotely via Telegram. I haven't gotten Telegram to work yet.

So far, this is sitting on my desk, I just have it set up today and will be testing with skittles or smarties.

Shaztech, 2024-09-05:
The problem with that is the "Multi-user" aspect of the project. If I make an "auto-throw" option it will interfer with that and mix pills into the chute. Also if the pills are not taken at the time of the alarm and multiple alarms occur (ex. the person is not home for the day), it will mix them all in the chute. It can be really dangerous.

The best option would be to add a large physical button on the side (or modify the base to fit it somewhere) and add a input pin to act as "Dispense"

[image]

burkby, 2024-09-06:
I like the idea of adding a physical button. I don't have the knowledge to do this but it's something that I will look into for sure. 

The auto drop idea has its drawbacks with potentially adding extra doses like you mentioned.

I don't use it for multi-user, it's a singular user in my scenario.

Also wanted to say thanks for making this device. It's a great piece of tech.

Shaztech, 2024-09-06:
I can add that to the code and show you where to solder the wire for the momentary switch signal + ground.

panthermatrix, 2024-09-06:
I made an easier solution myself since I am using ioBroker and other Smart Home controlls. The file **Web_Stuff.ino** from the source code here can also listen for external commands and the ESP32 board is very powerfull. So you can open an url on the pill dispenser and force drop out. This can be done with every web client like smartphone. For automation ioBroker or similiar software with script functionallity may help.

However, I recommend regularly checking whether the pill has been dispensed. There may be a mechanical defect, for example. It is therefore advisable to memorize the number on the label or note it in the ioBroker script or whatever and compare it after the drop-out. This way you can at least make sure that the pill has been dispensed. However, a mechanical button would have the same problems, as dementia can cause forgetfulness.

**Note:**
_A functionallity for forcing alarm sounds as a reminder on url call would be another idea, but I did not find the code for it yet._

**URL call is like:**
http://[ip_of_pill_dispenser]/droput?tray=[number_of_tray]

**Example:**
http://192.168.178.22/dropout?tray=1

Here is my code suggetion:
I added this at the start of "void webroute()" at line 415:

```
const int NUM_TRAYS = 10;

void webroute() {
  
  server.on("/dropout", HTTP_GET, [](AsyncWebServerRequest * request) {
    if (!request->hasParam("tray")) {
      request->send(400, "text/plain", "Param 'tray' is missing.");
      return;
    }
  
    int tray = request->getParam("tray")->value().toInt();
  
    if (tray < 1 || tray > NUM_TRAYS) {
      request->send(400, "text/plain", "Wrong value for tray.");
      return;
    }
      
   dispense(tray);
   request->send(200, "text/plain", "1");
 
  });

// the rest of the code ...
```

Shaztech, 2024-09-06:
Nice, thanks for that. I will also add it in the code as an option in the next update.

panthermatrix, 2024-09-06:
Perfect, thank you!

Perhaps this option can then be "External control via web service" or something similar.

I can think of two extensions, one is the one mentioned above and one is the alarm.

- http://[ip_of_pill_dispenser]/droput?tray=[number_of_tray]
Dropout pill from selected tray
- http://[ip_of_pill_dispenser]/alarm
Shows the alert screen „Take your pills“ with alert sound

**Idea for a button**
An external wireless button such as a Zigbee button from Ikea can be integrated into the local smart home network and either eject the pills when buttons are used (http://[ip_of_pill_dispenser]/droput?tray=[number_of_tray]) or delete an returning alarm (http://[ip_of_pill_dispenser]/alarm) if, for example, it is to be ensured that the pills are taken.

But this has nothing to do with the pill dispenser, it is an external script e.g. ioBroker. But the functionality to remotely control the pill dispenser would make this possible in the first place.

Shaztech, 2024-09-06:
I will probably add also this:
http://[ip_of_pill_dispenser]/droput?dispense

This command will drop the pills that are currently in alarm. And return an error if there's no active alarm.
So I can make a external button that replicate the "Dispense" touch, with a cheap ESP8266 and a button.

panthermatrix, 2024-09-06:
Good idea.

panthermatrix, 2024-09-07:
I have another suggestion. Sometimes during hot temperatures some trays may not work as expected. Here a /fix remote should be also implementet to fix stuck on half.

**Url can be like:**
http://[ip_of_pill_dispenser]/fix?tray=[tray]

**And my suggestion for /alam:**

**File: Dispense_Routine.ini:**
Add new method at the top:
```
void playAlarmSound(int id) {
  lv_timer_reset(alertsound_timer); // Reset alert sound timer
  Playsound(id); // Play the alarm sound
}
```

**File: Web_Stuff.ino**
Add between void webroute() { ... }:
```
  server.on("/alarm", HTTP_GET, [](AsyncWebServerRequest * request) {

    int alarmId = 2;
    if (request->hasParam("id")) {
      int alarmId = request->getParam("id")->value().toInt();
    }
  
   playAlarmSound(alarmId);
   request->send(200, "text/plain", "1");
  });
```

**You can use it like:**
- http://[ip_of_pill_dispenser]/alarm
This will play default alarm (number 2)
- http://[ip_of_pill_dispenser]/alarm?id=[custom_id]
This will play any sound beginning with the custom_id

Shaztech, 2024-09-07:
I was currently modifying the code. I've added the following:

http://[ip_of_pill_dispenser]/alarm    (Play the default alarm #1)
http://[ip_of_pill_dispenser]/alarm?id=[custom_id]      (Play the alarm #)
http://[ip_of_pill_dispenser]/dispense     (Dispense the current tray in alert, does nothing if there's no alarm)
http://[ip_of_pill_dispenser]/droput?tray=[number_of_tray]    (Drop the pill in tray #)
http://[ip_of_pill_dispenser]/dismiss       (Dismiss current alarm that are in progress)
http://[ip_of_pill_dispenser]/fix?tray=[number_of_tray]       (Make half step to fix tray #1 

GPIO4 can be pulled low to dispense the current tray in alert, does nothing if there's no alarm.

Fixed the typo in the tray config page: "Tray Pos. Adjustment:" -> "Tray Pos. Adjustement:"

I will upload the bin file here for you guys to test it.
Thanks again for your suggestions.

panthermatrix, 2024-09-07:
Thank you very much.

Shaztech, 2024-09-07:
Here is the beta version 1.7:

https://1drv.ms/u/s!AhrI-O28GqX7io8854s59nB6QsriPA?e=KMYMWu

Please try every "options" and tell me if it's ok !

panthermatrix, 2024-09-07:
I testet the PillDispenser.esp32.v1_7Beta.bin version via OTA. One thing does not work:

- http://[ip_of_pill_dispenser]/alarm (Play the default alarm no 1)
- http://[ip_of_pill_dispenser]/alarm?id=[custom_id] (Play the alarm "id")

**/alarm?id=[custom_id]** does not play a different alarm. If I type /alarm?id=3 or /alarm?id=4 it always plays the same alarm as /alarm does (filename: 2-xyz.mp3).

Can you please check if the **id** if set is recognized?

Shaztech, 2024-09-07:
It's supposed to work, I don't see anything wrong with the code.
Do you have more than 2 mp3 on the sd card ?
Does it play the first mp3 if you type /alarm?id=1 ? (Since the default alarmId is 2)

panthermatrix, 2024-09-07:
I have got 6 files on SD Card, all starting with the number. The both default files and:

- 1-Alert.mp3
- 2-PillsReady.mp3
- 3-Alarm1.mp3
- 4-Alarm2.mp3
- 5-Alarm3.mp3
- 6-Alarm4.mp3

but no matter what I enter as id, it will always 2-PillsReady.mp3 played.

I testet also:
/alarm?id=1

panthermatrix, 2024-09-07:
Now it works perfectly.

While testing the boot problem whith stuck on time sync #15 did not appear anymore.

Thank you very much.

closed 2024-09-07 by panthermatrix

Shaztech, 2024-09-07:
Did you test the with the physical input (GPIO4) ?

panthermatrix, 2024-09-08:
No, only the new remote functions yet. I can not test the physical inputs at the moment, since the pill dispenser is in use. But I will build another one as replacement and for testing soon.