Shaztech/Pilldispenser #15: Sometimes cold boot stucks at time sync ...

panthermatrix, 2024-09-07:
If I turn on the device after a while after a power disconnection and the time synchronization is activated, it sometimes happens that the boot process stops at "Syncing time..." stuks.

I was also able to understand this by a soft reset. Why this is the case is not entirely clear.

That's why I built in a second attempt as a solution and if this also fails, a reboot of the device. After the soft reboot, it has always worked so far.

But I could imagine that e.g. if there is no internet connection, a boot loop will be created here with my code suggestions. Do you have any idea how to avoid this? My suggestion here is to store a value like "reboot_on_timesync_fail = true" and to do without synchronization in case of failure despite soft reboot. A successfull rebot will set this value to reboot_on_timesync_fail = false.

Here's my code so far that works for me.

In file Loading_Screen.ino replace the void setupTime():

```
void setupTime() {

  int dsttemp;
  if (dstEnabled) {
    dsttemp = 3600;
  } else {
    dsttemp = 0;
  }
  configTime(utcOffset * 3600, dsttemp, "pool.ntp.org", "time.nist.gov");

  // Set time for first try
  unsigned long startTime = millis();
  unsigned long timeout = 10000; // Set to 10 seconds

  struct tm timeinfo;
  while (!getLocalTime(&timeinfo) && millis() - startTime < timeout) {
    Serial.print(".");
    delay(1000);
  }

  //
  int retryCount = 0;
  while (!getLocalTime(&timeinfo) && retryCount < 2) {
    Serial.println("Time synchronization failed. Try again...");
    delay(5000); // Wait for 5 seconds
    retryCount++;
  }

  // If all attempts have failed, perform a reset
  if (!getLocalTime(&timeinfo)) {
    Serial.println("Time synchronization failed. Perform a reset...");
    ESP.restart();
  } else {
    Serial.println("Time synchronized");

    // You can now use time functions like localtime_r to get current time
    char strftime_buf[64];
    strftime(strftime_buf, sizeof(strftime_buf), "%c", &timeinfo);
    Serial.print("Current time: ");
    Serial.println(strftime_buf);
    
    update_time_label(NULL);
    lv_timer_create(update_time_label, 15000, NULL); // Update every 15 Sec
  }
}
```

Shaztech, 2024-09-07:
I've never encounter that issue and I use my device daily. I've restarted it A LOT OF TIMES when I was testing and never had an issue with time sync. If there's no wifi, it just boot without clock. Maybe there's a problem with your acces point, or internet ?

panthermatrix, 2024-09-07:
Could be possible, but I have the problem even when I put the pill dispenser right next to the WLAN router. The wifi router uses WPA2/WPA3 and all other devices do not have problems connecting. The web ui of this devices works perfectly and never stucked. So wifi connection of this devices works. I also tried differend usb power plugs with more power 2.4A or iQ or other loading types.

1. The pill dispenser first connects to wifi -> successfull
2. pull dispenser shows correct IP
3. pull dispenser tries to sync time -> sometime it works, sometimes it does not work

Internet connection is working fine and also the time server are pingable.

This is kind of strange, but a soft-reboot always worked for me.

Shaztech, 2024-09-07:
I've changed the code like that:

  int dsttemp;
  if (dstEnabled) {
    dsttemp = 3600;
  } else {
    dsttemp = 0;
  }
  configTime(utcOffset * 3600, dsttemp, "pool.ntp.org", "time.nist.gov");

  // Wait until time is synchronized
  Serial.println("Synchronizing time");
  struct tm timeinfo;
  unsigned long startTime = millis();  // Record the start time

  while (!getLocalTime(&timeinfo)) {
    if (millis() - startTime > 30000) {  // Timeout after 30000 milliseconds (30 seconds)
      Serial.println("Time synchronization timed out.");
      break;
    }
    Serial.print(".");
    delay(1000);
  }

  if (getLocalTime(&timeinfo)) {
    Serial.println("Time synchronized");
  } else {
    Serial.println("Failed to synchronize time.");
    ESP.restart();
  }

I've made a 30 sec timeout. (it's safer that way)

If it cause any problem tell me. It will be with be with the bin "1.7" in the other issue

panthermatrix, 2024-09-08:
I think it works with this new code now. After several restarts the problem seems to be gone.
Thank you.

closed 2024-10-28 by Shaztech