Shaztech/Pilldispenser #18: Not an issue - just a question!

clucluclu, 2024-12-24:
Hi, this is a really impressive project!

I’ve been looking at how I can build an automated pill dispenser and integrate it with Home Assistant (popular home automation platform) to allow me to fire automations and alerts based on pill taken/not taken status, and refill readiness. Initially I was thinking easiest thing to do would be to use your hardware build, but use an esp32 running esphome instead and replace the logic entirely (mainly because I am familiar with coding for esp home and it integrates nicely with Home Assistant).

However, your code and project is so mature, that I’m wondering if I’m better off trying to simply make your build entirely and adding a few lines to the Arduino code so that it works as you intended, but also allows some information to be exposed to Home Assistant.

Various ways this could be achieved I suppose (and unsure if you or anyone else has ever considered same) - but before I give it a crack a do you have any advice/recomemndations you can share that might make it easier for me? -

- which variables will make the most sense for me to track?

- at which point in the codebase (ie which routine) would it make the most sense for me to call/embed my extra functions to ensure I get a real time feed as things change?

- is there is already some method baked in that allows for real-time access to this data over local network (prefer not to scrape the web interface if there is a way to get it more cleanly - though a simple REST style API would make it very easy.

- If you have any other good ideas or recommendations I’d love to hear them…you are already my hero for making such a polished project!

Thanks again - Clu.

Shaztech, 2024-12-24:
In the latest version I already added some actions that can be made with a web interface:

http://[ip_of_pill_dispenser]/alarm (Play the default alarm no 1)
http://[ip_of_pill_dispenser]/alarm?id=[custom_id] (Play the alarm "id")
http://[ip_of_pill_dispenser]/dispense (Dispense the current tray in alert, does nothing if there's no alarm)
http://[ip_of_pill_dispenser]/droput?tray=[number_of_tray] (Drop the pill in tray "number")
http://[ip_of_pill_dispenser]/dismiss (Dismiss current alarm that are in progress)
http://[ip_of_pill_dispenser]/fix?tray=[number_of_tray] (Make half step to fix tray "number")

for the variable to monitor, probably theses:

bool trayAlertEna[11] (is the tray is enabled for a daily alert)
bool traytriggered[11] (is the tray is currently in alert)
bool trayEnabled[11] (is the tray is enabled "installed")
bool traydisptoday[11] (is the tray has beed dispensed for today)
bool traydismtoday[11 (is the tray has been dismissed for today)

you can probably make your calls on the "Check_Alerts.ino" file. It's called every minute to check for alerts.
or you can modify "Web_Stuff.ino" and make requests for each info you need with HA via web requests.

no there's no method baked yet, but if you want to modify it, you can create a fork or include you modifications and I can add them in the next version ;)

I don't have HA yet, I'm still using an old version of Blynk with my own server (before they changed to subscription...)
I'm too lazy to change all my devices codes to switch (over 44 devices) XD

Thanks !

Shaztech, 2025-01-03:
check the v1.8 update, a lot of calls can be made:

Actions:

http://[ip_of_pill_dispenser]/alarm 						Play the default alarm no 1
http://[ip_of_pill_dispenser]/alarm?id=[custom_id] 	        Play the alarm "id" 
http://[ip_of_pill_dispenser]/dispense 					Dispense the current tray in alert, does nothing if there's no alarm
http://[ip_of_pill_dispenser]/droput?tray=[number_of_tray] 	Drop the pill in tray "number" (1 to 10)
http://[ip_of_pill_dispenser]/dismiss 					Dismiss current alarm that are in progress
http://[ip_of_pill_dispenser]/fix?tray=[number_of_tray] 		Make half step to fix tray "number" (1 to 10)

Informations:

http://[ip_of_pill_dispenser]/tr_name?id=[number_of_tray]	Return name of the tray
http://[ip_of_pill_dispenser]/tr_color?id=[number_of_tray]	Return 0 to 7 for assigned color (check color table)
http://[ip_of_pill_dispenser]/tr_inst?id=[number_of_tray]	Return 1 if tray is installed
http://[ip_of_pill_dispenser]/tr_enab?id=[number_of_tray]	Return 1 if tray is enabled for daily Alerts
http://[ip_of_pill_dispenser]/tr_trig?id=[number_of_tray]	Return 1 if tray is triggered (Alert in progress)
http://[ip_of_pill_dispenser]/tr_disp?id=[number_of_tray]	Return 1 if tray is dispensed for today
http://[ip_of_pill_dispenser]/tr_dism?id=[number_of_tray]	Return 1 if tray is dismissed for today
http://[ip_of_pill_dispenser]/tr_time?id=[number_of_tray]	Return alarm time for this tray

Color table:

0 = 255, 0, 0 = RED
1 = 255, 0, 255 = MAGENTA
2 = 128, 0, 128 = VIOLET
3 = 0, 0, 255 = BLUE
4 = 0, 255, 255 = CYAN
5 = 0, 255, 0 = GREEN
6 = 255, 255, 0 = YELLOW
7 = 255, 165, 0 = ORANGE

closed 2025-01-03 by Shaztech