Shaztech/Pilldispenser #20: Boards with Components Placed?

ssmithTaylor, 2025-01-02:
I'm a total noob in this area but I'd love to build one of these for myself, I'm terrible at remembering to take my medicine. I noticed some of the sites you can use to build the custom board will also do component placement as well, it does add to the expense but it doesn't seem too bad.

I'm not sure what the complexity of this ask is, but would it be possible to provide gerber files that already have the components placed so we can order them fully assembled?

Shaztech, 2025-01-02:
Check in the comments on Makerworld. There is someone that sell electronic
kits already done

jsondag, 2025-01-03:
That would be me. I have all the non printed parts available. Pick and choose whatever you like. 

Email me at 09-snakes.timbres@icloud.com and I can give details. 

This is my price list as of now. 

<p style="margin: 0.0px 0.0px 0.0px 0.0px; font: 12.0px Helvetica; min-height: 13.8px"><span style="font-family: 'Helvetica'; font-weight: normal; font-style: normal; font-size: 12.00px"></span><br></p>

Description | Quantity | Unit Price | Cost
-- | -- | -- | --
Main PCB, with components, smd assembly, loose pin header | 1 | $16.00 | $16.00
USB socket | 1 | $2.00 | $2.00
LED PCB, with components, smd assembly, loose pin header | 1 | $10.00 | $10.00
Tray PCB, with loose pin headers, straight through. | 10 | $1.00 | $10.00
ESP32 2.8” CYD display dev board. | 1 | $14.00 | $14.00
30mm 2W 8ohm Speaker | 1 | $2.00 | $2.00
DFPlayer mini | 1 | $3.00 | $3.00
Servo  SG90 (buy 1 per tray) | 10 | $2.50 | $25.00
P-Touch Number labels, White on Clear (buy 1 per tray) | 10 | $1.00 | $10.00
3mm OD left hand torsion spring (buy 1 per tray) | 10 | $1.50 | $15.00
Ballpoint pen springs x3 (buy 1 per tray) | 10 | $0.50 | $5.00
Tray Screws #2x5/16” (X10) #4x1/2” (x6). (buy 1 per tray) | 10 | $1.50 | $15.00
Base Screws #2x5/16” (X8) #4x1/2” (x8) | 1 | $1.50 | $1.50
Tig Rod (hinges) | 1 | $1.00 | $1.00
128MB micro SD card | 1 | $3.00 | $3.00
  |   |   |  
  |   |   |  
  |   |   |  
  |   | Subtotal | $132.50
  |   | US Shipping | $8.00
  |   | Total | $140.50

<p style="margin: 0.0px 0.0px 0.0px 0.0px; font: 12.0px Helvetica; min-height: 13.8px"><span style="font-family: 'Helvetica'; font-weight: normal; font-style: normal; font-size: 12.00px"></span><br></p>
<br class="Apple-interchange-newline">

Shaztech, 2025-01-03:
@jsondag Can I put this list and you email in the readme ?

jsondag, 2025-01-04:
Sure. No problem.

closed 2025-01-04 by Shaztech

ssmithTaylor, 2025-01-06:
@jsondag, I shot you an email but haven't received a response yet. Can you confirm the email above is correct? Maybe I got caught in your spam filter?

jsondag, 2025-01-06:
Strange. Could you tell me the day and time you sent the email?

ssmithTaylor, 2025-01-06:
Friday, January 3rd, 2025 at 12:42 AM ET

jsondag, 2025-01-07:
It was in fact in my spam filter. Along with three other inquiries. Unfortunately I get a ton of spam. I need to set up some rules to catch these emails. Thanks for bringing it to my attention.