Shaztech/Pilldispenser #21: Compile Error: Missing esp_int_widt.h in library AsyncElegantOTA-2.2.

mrperson2015, 2025-01-04:
I am new to ESP32/Arduino and developing on them. I am trying to setup the coding environment.  Meaning I could have this all wrong. 

**Issue**:
When I try to compile the current code I am getting an error about a missing file found as part of an include for library AsyncElegantOTA.h. Is this missing from checkin but found on your local environment? Maybe I have a wrong version of something?

**Error**: 
> Arduino: 1.8.18 (Windows 10), Board: "ESP32 Dev Module, Disabled, Disabled, Minimal (1.3MB APP/700KB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled, Disabled"
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
In file included from C:\source_code\Pilldispenser\Arduino\PillDispenser\PillDispenser.ino:13:
C:\source_code\Pilldispenser\Arduino\libraries\AsyncElegantOTA-2.2.7\src/AsyncElegantOTA.h:18:14: fatal error: esp_int_wdt.h: No such file or directory
   18 |     #include "esp_int_wdt.h"
      |              ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board ESP32 Dev Module.
This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

**Steps taken to setup environment**:
1. Cloned the repo
2. Created new branch
3. Installed Arduino IDE 1.8.18
4. Changed "Sketchbook Location" to [root]\Pilldispenser\Arduino
5. Added Additional Boards Manager URL: "https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json"
6. Changed board to "ESP32 Dev Module"
7. Changed "Partition Scheme" to "Minimal (1.3..."
8. Added tool for "ESP32 Sketch Data Upload"

Shaztech, 2025-01-05:
Try to downgrate your esp32 version to 2.0.17
[image]

mrperson2015, 2025-01-05:
What did I do wrong that the version you suggested isnt in my list? I have tried many other versions though w/ no success.
[image]

Shaztech, 2025-01-05:
[image]

mrperson2015, 2025-01-05:
You are FAST. I had been filtering it to ESP32 in the Type dropdown. That doesn't show the esp32 by Espressif Systems. Ugh. I have the correct one selected and installed. Compiling now. It is taking longer than it was ;)

mrperson2015, 2025-01-05:
WORKED!!! Thank you

closed 2025-01-05 by mrperson2015