Shaztech/Pilldispenser #22: Servos wont turn to dispense pills

walkeryan, 2025-02-19:
I have everything populated and wired and programmed, my led flashes the correct color but when I try to do anything related to the servos I get nothing. When I plug the servo in it jumps a tiny bit and will self center if I do that repeatedly but that's the only action I get out of it. I've went over all of my components and connections and even reflashed. Any ideas where I can go from here?

Shaztech, 2025-02-19:
Try another USB power supply, check that the wiring for the SCL, SDA are on
the right pins (the colors of the cable that came with your board can be
inverted).

walkeryan, 2025-02-19:
Thanks, I've checked the SCL, SDA and they are correct, also tried 3 different power supplies. I went with the straight through boards and have bridged the first pads in the middle, I'm assuming thats correct but I just want to verify? Nothing in the write up about it. Thanks!

walkeryan, 2025-02-19:
Also, just discovered that the servo isn't centering when I apply power, its just rotating counterclockwise about 1/16th of a turn until it gets the the end and locks up?

Shaztech, 2025-02-20:
The servo just get power that's why is not centering

[image]

Check the straight board. You need to bridge only one for the tray you want (white mark is for tray #1 next is #2). Each tray need to be unique.

Shaztech, 2025-03-01:
I assume your problem is fixed.

closed 2025-03-01 by Shaztech

Chipper2, 2025-03-23:
I'm having the same issue.  I wanted to do some testing before putting it all together because my soldering skills are very lacking.  To the point I messed up the USBC plug but was able to use the power part of the plug. Are the data wires needed if you are only using them for power?

The display works great but when I plug a motor up, nothing moves.  I swapped the SCL, SDA and there was no change.

I do have a question about the straight boards.  I see above that we need to bridge the appropriate span for the trey that it is installed into.  I took a volt meter to test continuity, and as I tested one by one, all of the corresponding pins were connected.  I may not understand the process of making those work.

Thank you for your time and invention.
Chip