Shaztech/Pilldispenser #23: Request for BOM or Component List for JLCPCB

Joshicopter, 2025-02-22:
Hi Shaztech,

First of all, thanks for your great project! I'd love to build it and have the PCB manufactured by JLCPCB. However, I’m not very experienced with soldering, so I plan to order the PCB from JLCPCB directly. To make the process easier, it would be really helpful if a BOM (Bill of Materials) or other compatible component lists for JLCPCB were available. 

Would it be possible for you to provide these files? That would make ordering much more convenient for me (and probably for others as well).

Thanks in advance!

Shaztech, 2025-02-25:
I don't have theses files. You need to create them.

closed 2025-02-25 by Shaztech

jsondag, 2025-02-25:
I have fully populated boards available. Along with all of the non printed parts. 

Shoot me an email and I’ll give you all the details. 

09-snakes.timbres@icloud.com

jsondag, 2025-04-01:
Apologies. This seems to have been caught by my spam filter. I thought I’d excluded keywords, but apparently spam filters run before the rules I’d set. Please let me know if you still didn’t get a reply, as I had a couple in there and I’m not sure which might be you.