Shaztech/Pilldispenser #26: Can't seem to get motors to turn.

Chipper2, 2025-03-23:
I'm having the same issue. I wanted to do some testing before putting it all together because my soldering skills are very lacking. To the point I messed up the USBC plug but was able to use the power part of the plug. Are the data wires needed if you are only using them for power?

The display works great but when I plug a motor up, nothing moves. I swapped the SCL, SDA and there was no change.

I do have a question about the straight boards. I see above that we need to bridge the appropriate span for the trey that it is installed into. I took a volt meter to test continuity, and as I tested one by one, all of the corresponding pins were connected. I may not understand the process of making those work.

Thank you for your time and invention.
Chip

Chipper2, 2025-03-23:
I have also been tring to work on this.  But i'm on a Mac and aren't having much luck.

The first Firmware Installation:

Download the 5 bin files in this release
Download the esptool.exe from here https://github.com/espressif/esptool/releases/tag/v4.7.0
Copy the 5 bin files to the root of the folder (with esptool.exe)
Connect your device (the ESP32 module with the LCD) to your computer via USB
Find the Serial COM port number (in device manager)
open a command line inside your folder and type, change the COM port that match your device (don't forget to change the files version ex: v1.3):
esptool.exe --chip esp32 --port COM4 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x1000 PillDispenser.bootloader.v1_3.bin 0x8000 PillDispenser.partitions.v1_3.bin 0xe000 boot_app0.bin 0x10000 PillDispenser.esp32.v1_3.bin 0x3D0000 PillDispenser.spiffs.v1_3.bin

Shaztech, 2025-03-23:
- The USB C data line is for programming, they are optionnal but you need to program it first with the screen outside the assembly.
- For the servos, there's a version of these that are Digital and not PWM (Analog), You need to be sure they are the right type.
- The straight board version need to have ONE solder bridge on each tray like so (the white mark), the first bridge is for the 1st tray:

[image]

- I don't own a Mac, and the procedure is probably different. (port number, you can't run an exe on mac, the command may be different...) Here's a example from chatGPT:

```
install esptool: pip install esptool
list serial: ls /dev/tty.*
Replace /dev/tty.usbserial-xxxxx with the actual port.

esptool.py --chip esp32 --port /dev/tty.usbserial-xxxxx --baud 921600 --before default_reset --after hard_reset write_flash -z \
  --flash_mode dio --flash_freq 80m --flash_size 4MB \
  0x1000 PillDispenser.bootloader.v1_3.bin \
  0x8000 PillDispenser.partitions.v1_3.bin \
  0xe000 boot_app0.bin \
  0x10000 PillDispenser.esp32.v1_3.bin \
  0x3D0000 PillDispenser.spiffs.v1_3.bin
```

Chipper2, 2025-03-23:
I think I figured out the motor part.  Looks like a looks connection.

As far as programming.  I can't seem to get it to connect to a COM port.  I'm fairly certain I have messed up the chip on the back while soldering.  However, it does seem to connect via wifi, and I'm able to program it from there.

What do the new files do?  Is it imperative that I do this part?  If so, I'll probably need to order a new screen.

Thank you for your time.
Chip

Chipper2, 2025-03-23:
New symptom: if I plug in the first motor, it works great, but any motor that I connect after that doesn't work.  If I move #2 down to the bottom, it works, but nothing after that does.  I have tried all motors individually, and they work, but if I stack them, they don't.

Shaztech, 2025-03-23:
If you have programmed at least once and you have access to the firmware update web interface. The only files you need are 
PillDispenser.esp32.v1_8.bin
PillDispenser.spiffs.v1_8.bin
That's the lastest version.

Maybe your powersupply that you use is not strong enough. Or check that your solder points are ok on the top part of the board on the tray

Chipper2, 2025-03-23:
Thank you, yes, it's once again my soldering skills.  Thank you for giving us such a wide section to shoulder.  To the guys that made the controller, you suck!!  No insult intended.  LOL

Thank you for letting me know about the updating over the air.

Chip

Shaztech, 2025-03-23:
When you talk about the controller you mean the screen module ?

Chipper2, 2025-03-23:
Yes, sir, the screen module.  I broke the LED connections trying to get that off.  On the USB C connector, I broke the D+ and D-.  I didn't even try the speaker, lol.  I drilled a hole in the side of the base to gain access to the USB-C to program it and power it, but then I realized I think the power for the motors comes from the external USB-C port.

Obviously, this is no wear your fault, or anyone else's for that matter.  I suck at soldering, and I knew I did, but I took the risk anyway.  For the most part, I have it working, only issue I'm having now is it doesn't advance as well as I thought it should.

[image]

Chipper2, 2025-03-23:
https://github.com/user-attachments/assets/dc3dc41f-e6bf-4138-8d82-023f19e7502f

Shaztech, 2025-03-23:
You need to feed the servo board 5v, you don't have a choice.
The tray should turn without freely (when the lock and motor are not installed). If it's too tight maybe scale it down to 99.5% or something like this. But I didn't had any complain on that.

Shaztech, 2025-03-23:
Try to reprint the arm, there's goop at the base, also does the lock is printed and in place on the other side ? I don't see there

Chipper2, 2025-03-23:
Chip Thomas

Chipper2, 2025-03-23:
I will reprint.

Thank you so much for the customer service!!!
Chip Thomas

Chipper2, 2025-03-23:
I reprited the push art.  But its like the nub on the end of the motor is't long/big enough.  Am I suspose to print a different one?

Shaztech, 2025-03-25:
I've printed another tray for myself. And tested the Arm V2 (this arm was designed by another user) and it's not working... I will work on that and update the file. I will tell you when it's done.

Chipper2, 2025-03-25:
No problem.  I actually went back and printed the original that you did.  The one that has several parts seems to be working well.  thank you for letting me know.

Shaztech, 2025-03-25:
Yeah the original is reliable. I have 5 trays with it, and use it daily for the last year plus. No problem, more parts but reliable.

Chipper2, 2025-03-26:
I didn’t mean to get the V2 arm banned. lol

Is there a way to get the esp32 and touch scream separate?  I’m still having problems with my soldering.

Shaztech, 2025-03-26:
I've tried to fix the V2 arm (that I never tested before) for 5 hours, like 8 prints differents. And it wasn't working... 
I gave up, printed my original version, first try and it was working perfectly. I don't want people to have something that is not reliable in a pill dispenser that is crucial.

For the esp32 and touch screen, you can buy parts separate and find the schematic to connect everything at the same place on the esp. But it's just MORE (a lot more) soldering.

Just contact [jsondag](https://github.com/jsondag) is selling kits, you can email him at: [09-snakes.timbres@icloud.com](mailto:09-snakes.timbres@icloud.com). He will help you.

Shaztech, 2025-03-26:
Also if you mess some pads where the led was, you can find them directly on the ESP32, check a pinout of the module.